echostack-org/echoregions #16: Finding files based on metadata of the sonar files.

valentina-s, 2021-06-25:
Currently, finding the sonar files of echoregions relies on the data in the name of the files. This is fine for our hake survey use case, but it eventually we should make it work with the metadata from the sonar file (opened without specifying any group), so that it is file name independent.

leewujung, 2023-05-22:
The sonar Sv files do not have specific groups since they are standard xarray datasets. What's needed here is to use the coordinate `ping_time` in these datasets to select which files to use.

We should make clear that we encourage people to consolidate data from many .raw files into mid-size (e.g., transect-length) EchoData or Sv .zarr stores, but not gigantic stores, since otherwise it will take forever for the coordinates to load.

leewujung, 2024-03-26:
I think this was addressed in #169.

closed 2024-03-26 by leewujung