echostack-org/echoregions #42: Region Masking Functionality

valentina-s, 2022-05-10:
The `mask` function of the `Regions2D` object is working currently. 

The actual functionality is in the `mask` method of the [`Regions2DMasker`](https://github.com/OSOceanAcoustics/echoregions/blob/main/echoregions/mask/region_mask.py)

Working example for making "Hake" region masks (only one region) in this [notebook](https://github.com/uw-echospace/hake-labels/blob/main/notebooks/convert_region_to_mask_all.ipynb).

Should we store more than the original `id`s in the masks? We can always separately store all label region ids with some metadata as a separate dataframe, or make it part of the mask dataset.

A few extra details:

[] adding regionmask version as a requirement
[] adding __init__.py
[] add examples in notebooks
[] add examples selecting different regions based on labels

leewujung, 2022-12-06:
#45 partially addressed this issue.

valentina-s, 2023-05-26:
The only one that is not addressed is 3) storing different regions in different layers. I will make a stand alone issue and we can discuss the format. #75

closed 2023-05-26 by valentina-s