echostack-org/echoregions #43: Line Masking Functionality

valentina-s, 2022-05-10:
Making bottom masks from `.evl` files requires to interpolate the bottom curve points to the grid of the sonar file. The choice of the interpolation scheme affects the result. We should provide option to select different interpolation schemes, and the user can change the parameters so this works for different datasets.

Version for Hake Survey is in this [notebook](https://github.com/uw-echospace/hake-labels/blob/main/notebooks/bottom_processing/convert_bottom_from_csv_to_mask_all_exports_new2021nov.ipynb)

leewujung, 2023-05-26:
@valentina-s will add one more example to this from x22.

leewujung, 2023-05-26:
Also need to revisit the mask formats.

valentina-s, 2023-06-02:
Here is an example [notebook](https://github.com/uw-echospace/hake-labels/blob/2019/notebooks/examples/Bottom_for_x0022_2019.ipynb) making a mask for transect x22  from 2019 survey.

valentina-s, 2023-06-02:
In this [notebook](https://colab.research.google.com/drive/1SVLNIc5fDACICOZ16dTzbuJkZS78XgDu) there is a more in depth inspection of the effect of the interpolation schemes.

closed 2023-07-02 by ctuguinay