echostack-org/echoregions #58: Revamp testing suite with new converted file from echopype 0.6.3

valentina-s, 2022-12-19:
- [ ] potentially use fixtures where possible
- [ ] ensure testing with raw files, nc, zarr files
- [ ] make sure only small testing files are in github

ctuguinay, 2023-06-29:
Do we still want to test on raw and zarr files? I was under the impression we were going to only work with .nc files.

ctuguinay, 2023-06-30:
This issue should be addressed in the same PR that addresses #107.

valentina-s, 2023-07-03:
[Here](https://drive.google.com/drive/u/0/folders/1XKtwlPDD7tAVPswf4GwoUVaKItgSa9K9) is a link to a zip file with a more recently converted example from 2019 Survey. It also has a bottom if needed. It is a whole transect but it can be cut into pieces/and possibly further subsampled if needed. I am assuming you want to test with an example which already has depth and regular grid incorporated? But could provide also some more raw files in `.zarr` format which have not been yet combined.

valentina-s, 2023-07-03:
> 

So the functions should be tested on an xarray (which may have come from `.zarr` or `.nc`), and for simplicity we can make the reading from `.nc`. I think the only difference that can come from the different formats is the way they are chunked, and whether they are loaded into RAM or not. I assume if they are not the `regionmask` function will invoke it? We have converted `.zarr` files handy, and I think opening them with `xarray` and converting them to `.nc` will be the same as converting them from scratch.

valentina-s, 2023-07-03:
> Do we still want to test on raw and zarr files? I was under the impression we were going to only work with .nc files.

So the functions should be tested on an xarray (which may have come from .zarr or .nc), and for simplicity we can make the reading from .nc. I think the only difference that can come from the different formats is the way they are chunked, and whether they are loaded into RAM or not. I assume if they are not the regionmask function will invoke it? We have converted .zarr files handy, and I think opening them with xarray and converting them to .nc will be the same as converting them from scratch.

commit 36d7eaf6 (2023-08-02): Improve Testing (#115)

## Improve Testing

This PR closes #58, #107, #85, #112, #113, #99, #53, #28, #120:

- Created and used a new testing suite: New EVL, new EVR, new Zarr file
- New testing suite resolved the warning that arose from #85
- Refactored tests
- Added fixtures
- Added hints
- Added marks
- Added/improved upon numpy docstrings for tests and fixtures
- Added further tests for Lines and Regions2D
- Added functionality and tests for EVR parser to handle missing bbox values and no regions
- Removed the unused convert_points function
- Fixes docstrings that appeared as errors in the [... cut]
closed 2023-08-02 by ctuguinay