echostack-org/echoregions #72: Add functionality to create masks for within-transect (good regions)

valentina-s, 2023-05-22:
This can take several forms but it slightly depends on how people have annotated individual breakpoints:
* start transect
* break transect
* resume transect
* end transect

Usually those should be annotated with log lines, but sometimes it can be thin boxes, or `within-transect` boxes. One approach is to create an intermediate table with time stamps for the staring and ending times and create the mask based on it.

leewujung, 2023-05-22:
I wonder if this should be something like an external script we use for the hake survey data with the start/end times, it seems cleaner to have it separate from the main echoregions functions.

valentina-s, 2023-05-22:
Right now it is external, but we should possibly have some functionality to handle log-type regions in `echoregions` since this is one of the types of regions.