echostack-org/echoregions #82: Interpolation for Line Plot

ctuguinay, 2023-05-30:
A possible feature could be using interpolation to connect some of these dots. More specifically, the interpolation algorithm would have to generate associated timestamp and depth values.

[image]

valentina-s, 2023-06-02:
Yes, in fact interpolation is need to create a bottom mask aligned with the Sv data (the bottom grid is interpolated on the Sv time grid). See #43

ctuguinay, 2023-06-02:
Potentially use the following interpolation scheme [https://github.com/uw-echospace/hake-labels/blob/2019/notebooks/examples/Bottom_for_x0022_2019.ipynb](https://github.com/uw-echospace/hake-labels/blob/2019/notebooks/examples/Bottom_for_x0022_2019.ipynb)

closed 2023-07-02 by ctuguinay