echostack-org/echoregions #106: Update Notebooks (After New Functionality Added)

ctuguinay, 2023-06-29:
Due to a number of new functions being added, and paired with refactoring, there exist major discrepancies between the examples in the Notebooks and how the current package functions.

The following tasks address this issue:

- [ ] Create `Lines_functions.ipynb`
- [ ] Create `Lines_masking.ipynb`
- [ ] Update `Lines_plotting.ipynb`
- [ ] Update `Regions2D_functions.ipynb`
- [ ] Update `Regions2D_masking.ipynb`
- [ ] Update `Regions2D_plotting.ipynb`

ctuguinay, 2023-07-04:
Also, since PR #114 is removing the `requirements.txt` folder for notebooks, it is necessary to add something regarding notebook requirements.

Notebook requirements are as follows:
`dask
xarray>=2023.2.0
echoregions @ git+https://github.com/ctuguinay/echoregions@65a784fc5c9c86f9bc19ea91cbed6e3422554954
gdown`

leewujung, 2023-11-25:
@ctuguinay : All of the tasks were addressed prior to the v0.1.0 release, right? If so, could you close this so we can complete 100% the first release milestone?

ctuguinay, 2024-04-10:
Closed with the merge of #184

closed 2024-04-10 by ctuguinay