echostack-org/echoregions #116: Add EchoRegions to Package Repositories

ctuguinay, 2023-07-07:
After the first release of EchoRegions, it should be added to the following popular package repositories:

- [ ] PyPi
- [ ] Conda
- [ ] Zenodo (for the DOI)

To help facilitate this addition of EchoRegions to the previously mentioned package repositories, one should follow the code detailed in Echopype.

leewujung, 2023-09-24:
Also see #68 for reference. Feel free to close that one if everything is done/captured here.

leewujung, 2023-11-25:
Seems like we just need to add this to conda. I'll move this to under v0.1.1 release.

leewujung, 2023-12-22:
Since `echoregions` is now on pypi and zenodo, we can do without conda for now. I'll close this issue and open a new one for adding conda recipe.

closed 2023-12-22 by leewujung