echostack-org/echoregions #117: Allow Option in Regions2d Mask Conversion for Label Names

ctuguinay, 2023-07-10:
Currently, Regions2d Mask Conversion functions only allow integer label names in both the 2d and 3d masks. For fish labeled .evr files, it would be beneficial to have the option to have label names in both the 2d and 3d masks that match the region names, like "Hake" and "Rockfish"; so in essence, we want to add the option to have a dictionary that maps string label names to integer label names in the 2d mask.

valentina-s, 2023-07-14:
Some transects from 2017 survey with different labels which could be used to fiddle with different labels.

x100: Herring: 4, Other 3, Hake: 1

x10: Hake: 9, Age 1: 3

leewujung, 2023-12-11:
I think this issue is the same as #131. @ctuguinay : could you confirm?

ctuguinay, 2023-12-22:
This issue is too specific behavior-wise for a general package, and this behavior can already be accommodated by the more flexible `mask_labels` dictionary. The user just needs to create the mapping between `region_id` and label names, and set that as the `mask_labels` input.

closed 2023-12-22 by ctuguinay