echostack-org/echoregions #125: Add Clean Times and Add Time Conversion

ctuguinay, 2023-09-20:
- [ ] Implement clean times from Hake Labels into utils. Ensure that it works for both Lines and Region2D objects.
- [ ] Implement add time conversion from Hake Labels into utils. Ensure that it works for both Lines and Region2D objects.
- [ ] Write clean times and add time conversion tests

ctuguinay, 2023-12-11:
@leewujung Now that I'm thinking about this, this function (https://github.com/uw-echospace/hake-labels/blob/e63b392287589f1eb8878e44dd0919040eb8b16d/hake_labels/label_allocations/utils_transects.py#L17) in its entirety seems a bit too specific for Echoregions.

How about instead of this, I have a little helper function to Regions2d for adding PST times?

leewujung, 2023-12-11:
Looking at this again I am not sure what the "cleaning" part refers to...?

> How about instead of this, I have a little helper function to Regions2d for adding PST times?

Sounds good. How about having this function to add a column of time from UTC to some user-specific time zone? So that it can work for PST and other times. It feels a little dumb though, since it is a pandas function. Maybe we just don't add anything and leave it to the users to do whatever they need to do.

ctuguinay, 2023-12-11:
Oh true, it does seem very user specific. Let's close this and let the user decide what they want.

closed 2023-12-12 by leewujung