echostack-org/echoregions #223: No module named '_echoregions_version'

oftfrfbf, 2025-02-06:
Trying to run any tests there is a problem with the file "_echoregions_version.py" missing:
```python
from _echoregions_version import version as __version__  # noqa
E   ModuleNotFoundError: No module named '_echoregions_version'
```

I forked the code and am currently looking at a way to fix this.

leewujung, 2025-02-06:
This is defined in `pyproject.toml` and autogenerated. You should't have to generate it.

oftfrfbf, 2025-02-06:
Ah beat me to it. I wasn't running the command from the documentation properly. Where is say here <https://echoregions.readthedocs.io/en/latest/installation.html>:
```
pip install -e
```
it should be
```
pip install -e .
```

What had initially led me here was trying to use the echoregions library as a third party. Installing it was giving the same problem with a missing version. I'm wondering if the version metadata was somehow missing when the package was last built? (or maybe I'm doing things wrong again?)
[image]

oftfrfbf, 2025-02-06:
Making the change now, will submit it soon.

I'm changing the paths to the test resources too if that's okay, they don't work on a windows system the way they are currently configured. Specifying them the same way as they are in echopype.

leewujung, 2025-02-06:
Sounds great. Thank you!